Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #1: Setting the Low Datarate Optimization flag

LoRaTracker, 2018-01-12:
I have looked at your code and I cannot see where the LowDataRateOptimize flag is being set, or can be set ? 

See section 4.1.1.6 of the SX127x datasheet, the flag (bit 3 of reg 0x26) needs to be set when the symbol time exceeds 16ms. 

I can describe the symptoms of not setting the flag if you want ..........

Wei1234c, 2018-01-12:
What's the symptoms ?
Can I leave it on even  when the symbol time does not exceeds 16ms ?

Thanks !

LoRaTracker, 2018-01-13:
The symptoms of not setting the optimization flag correctly are 
corruptly packets at higher spreading factors, the problem gets worse 
the more power you use.

For reliable operation the flag needs to be set as per the data sheet, 
off for symbol time below 16mS, on for symbol time more than 16mS.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Wei1234c, 2018-01-13:
The driver already provides "readRegister" and "writeRegister" methods, with which you can do whatever you want. 

For your purpose, for example:

controller = config_lora.Controller()                       
lora = controller.add_transceiver(sx127x.SX127x(name = 'LoRa'),
				  pin_id_ss = config_lora.Controller.PIN_ID_FOR_LORA_SS,
                                  pin_id_RxDone = config_lora.Controller.PIN_ID_FOR_LORA_DIO0)

reg_val = lora.readRegister(0x26)
lora.writeRegister(0x26, reg_val | 0x08)   # enable LowDataRateOptimize
lora.writeRegister(0x26, reg_val & 0xF7)  # disable LowDataRateOptimize

LoRaTracker, 2018-01-13:
I know you can set the flag with those methods, but I suspect the vast 
majority of users of the library  will not realise this required flag  
even exists.

There are methods in the Library for setting bandwidth and spreading 
factor,  a lot of people might then assume that is the setup complete, 
and its not as the optimization flag also has to be set.

You can calculate the symbol time from the current bandwidth and 
spreading factors so the flag can be set automatically.

A compromise is to have a separate method for setting the optimization 
flag and a note as to when it should be used, you can use the Semtech 
LoRa  Calculator which will calculate the symbol time for you.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Wei1234c, 2018-01-15:
I add following lines in sx127x.py, it should do the job.

```
        # set LowDataRateOptimize flag if symbol time > 16ms (default disable on reset)
        if 1000 / (self.parameters['signal_bandwidth'] / 2**self.parameters['spreading_factor']) > 16:
            self.writeRegister(REG_MODEM_CONFIG_3, 0x04 | 0x08)
```

closed 2018-01-18 by Wei1234c