Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #2: TTGO range and data rate

carterw, 2018-01-13:
Wei1234c thanks very much for your excellent repository of code here and also the reference material! Very impressive work.

We are discussing the sx127x driver code in [this thread on the Micropython forum](https://forum.micropython.org/viewtopic.php?f=18&t=4196), you may be interested in it.

Seeing as how you have code that references the TTGO modules, have you experimented with them? I am curious to know what distances and data rates can be expected.

Wei1234c, 2018-01-14:
Glad that it is useful to you.
As for the TTGO ESP32+LoRa+OLED module, I have used it in some experiments but not in field test of long range.
I think that there are mainly two problems of TTGO module:

1. Hardware reset sometimes cause problem: I think TTGO module has bad reset circuit so that the SPI sometimes doesn't work properly after hardware reset. Everything works fine after software reset though. This only happened to TTGO module, my other ESP32 modules work just fine with SX1278 modules..
2. Power consumption: If you set transmitting power level high for the SX1278, it will require as many as 120mA. The regulator IC may not be able to supply it along with the consumption of other components on the development board.

carterw, 2018-01-14:
The default transmit power level is 2. Is that the highest? I have tried powering with a lipo battery, which probably bypasses the regulator, and not seeing any difference.

Wei1234c, 2018-01-14:
The pwr_level can go up to 17.
I am not sure about LIPO, which I didn't use.

Wei1234c, 2018-01-14:
Also, you may want to try setting the LowDataRateOptimize flag, please see issue #1.

carterw, 2018-01-14:
Adjusting the power level to 17 made a big improvement in the range, thanks.

closed 2018-01-14 by carterw