Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #5: Transmission with Ack

rahulrksmau, 2018-05-02:
How can I send msg with Ack that msg has been received ?

Wei1234c, 2018-05-02:
You will need to implement your own handshake or protocol for communication with acknowledgement.

Or you can use some other LoRa hardware and driver which follow the specification of LoRaWAN and may have ACK capability built-in.

rahulrksmau, 2018-05-02:
Ok.
One more thing ,
In Codes/demo/controller_rpi.py  you have declared ON_BOARD_LED_PIN_ON = 47 
how is this possible?

Wei1234c, 2018-05-02:
Somehow the tiny green LED on Raspberry Pi Zero W is connected with pin 47.

rahulrksmau, 2018-05-03:
Do you have connection diagram or picture of  LoRa with RPi ?

Wei1234c, 2018-05-03:
https://2.bp.blogspot.com/-mFJmFVdC8RI/Wa0K1Q38ZeI/AAAAAAACPhk/hlcWCT3qrIouCTd6KIHJpIk4rCtW83ApwCLcBGAs/s400/RPi_new.jpg.jpg

https://wei1234c.blogspot.tw/2017/08/sx127x-lora-transceiver-driver-for.html

this is irrelevant to the topic of this issue. close it now.

closed 2018-05-03 by Wei1234c