Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #10: Information please

johncblacker, 2019-02-19:
1.) Can you provide me with information on how to install this driver code.
2.) What is in the "tools" directory?
3.) Is there any configuration required for ttgo?

Thank you

johncblacker, 2019-02-21:
Is anybody monitoring this repositories issues?  Seems to be dead.

Wei1234c, 2019-02-22:
1.) please refer to : https://forum.micropython.org/viewtopic.php?f=16&t=3871  and  https://wei1234c.blogspot.tw/2017/08/sx127x-lora-transceiver-driver-for.html
2.) "tools" folder contains Jupyter notebook to upload uPy file to ESP8266/ESP32.
3.) please look into "controller_esp_ttgo_lora_oled.py".
4.) please also refer to: https://forum.micropython.org/viewtopic.php?f=16&t=3871#p33901

closed 2019-02-22 by Wei1234c