Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #12: spi and i2c problem!

monster-xie, 2019-04-20:
I use the esp32 with oled board. pin define same as your code,but the spi for lora and i2c for oled don't work.

Wei1234c, 2019-04-20:
The SPI communication often failed on the OLED ESP32 DevKit. It fails after h/w reset until s/w reset.
Try soft reset (ctrl-D) it.

monster-xie, 2019-04-20:
when ctrl-D，then get the picture below，oled no display？

[image]

Wei1234c, 2019-04-20:
Please trace the "config_lora.py" file to see whether the right controller_xx.py is selected.

closed 2019-04-20 by Wei1234c

monster-xie, 2019-04-21:
controller_esp_ttgo_lora_oled imported but not display!

Wei1234c, 2019-04-22:
I just flash and re-upload the code. It works perfectly on my ESP32-LoRa-OLED DevKit. 
Maybe the OLED display on your borard is out of order.