Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #13: LoRaWAN OTAA and ABP?

beyonlo, 2019-11-25:
Hi, this driver works with the LoRaWAN using the OTAA and ABP activation?

I need to use my ESP32 device with LoRa using a LoRa provider (LoraWAN), is possible?

Thank you.

Wei1234c, 2019-11-26:
No, this driver doesn't follow LoRaWAN's standard.

However, you might want to consider [MQTT bridged LoRa networks](https://github.com/Wei1234c/MQTT_bridged_LoRa_networks_with_MicroPython_on_ESP32).

closed 2019-11-26 by Wei1234c

carterw, 2019-11-26:
It is a low-level driver that directly sends and receives arbitrary data, there is no network protocol on top of it.

beyonlo, 2019-11-27:
@carterw yes, you are right.

Do you know any project that implement the LoRaWAN standard with OTAA/ABP activation on MicroPython for ESP32 using the SX127X?

Thanks