Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #14: After awakening from deep sleep mode - ESP32

itsnils, 2019-12-13:
Very great project😉 simple and short 👍 

I can't get any further at one point😢
I put the ESP32 into deep sleep mode. when it wakes up, no more data will be transferred. does anyone have an idea? or is there a clean guide how to do after deep sleep?

Thank you for your great work😊

carterw, 2019-12-13:
When the ESP32 wakes up after deep sleep it is as if it were reset, nothing remains of the previous boot state. You have to run all your startup code including the driver.

itsnils, 2019-12-13:
thanks for the quick help😀

I took into account that everything will be reset.(boot.py and end programmed accordingly). when i connect the ESP 32 to power and disconnect and reconnect it all works fine. but when it opens from the deepsleep then the ESP32 works as desired.
I have the feeling that I have to send a reset to the SX127x because the registers still have the old values because the SX127x is not disconnected from the voltage when the ESP 32 goes into the deepsleep.

Do you have another idea or solution?

carterw, 2019-12-13:
Yes I think you do have to toggle the reset on the SX127x.

On a project I've done I wanted the ESP32 to hibernate in deep sleep on battery power but could not get the SX127x to go into low power mode. I had to put a mosfet on the SX127x power circuit that would turn off power to it when the ESP32 did not enable it.

Wei1234c, 2019-12-13:
to toggle reset pin (to all attached transceivers):
controller.reset_pin(controller.pin_reset)

itsnils, 2019-12-19:
cool it works😍 thanks for the support😊

closed 2019-12-19 by itsnils