Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #15: I need help

thanapol199717, 2021-04-19:
I want to use raspberry pi as gateway but can reply to my node where my node is using arduino nano I can get data from node normally but raspberry pi I can't send data back to node. I've been trying for several days. Hope someone can help me.

I am using a Lora sx1276 module.

I saw you on youtube, so I message to you.
Thank you if you can help me it will be very helpful. Because I am very weak with this.

Wei1234c, 2021-04-20:
Please state your configuration, obsevation more expressively and specifically.

It's hard to tell it is actually an "issue" without detailed information. Close for now.

thanapol199717, 2021-04-20:
My problem is that the Raspberrypi cannot send data back to the Arduino. But Arduino can send data to Raspberrypi normally. I don't know if it is in my programming or at hardware.

thanapol199717, 2021-04-20:
This is my code at raspberrypi.

from time import sleep
from SX127x.LoRa import *
from SX127x.board_config import BOARD
import board
import datetime
import sys
import time
from influxdb import InfluxDBClient

BOARD.setup()
host = "192.168.1.231" # My Ubuntu NUC
port = 8086 # default port
user = "mushroom" # the user/password created for the pi, with write access
password = "mushroom"
dbname = "mushroom" # the database we created earlier
interval = 60 # Sample period in seconds

client = InfluxDBClient(host, port, user, password, dbname)

# think of measurement as a SQL table, it's not...but...
measurement = "mushroom1"
# location will be used as a grouping tag later
location = "Mushroom"

class LoRaRcvCont(LoRa):
    def __init__(self, verbose=False):
        super(LoRaRcvCont, self).__init__(verbose)
        self.set_mode(MODE.SLEEP)
        self.set_dio_mapping([0] * 6)

    def start(self):
        self.reset_ptr_rx()
        
        self.tx_counter = 0
        self.write_payload([255, 255, 0, 0, 104, 101, 108, 108, 111])
        
        self.set_mode(MODE.RXCONT)
        while True:
            sleep(.5)
            rssi_value = self.get_rssi_value()
            status = self.get_modem_status()
            sys.stdout.flush()
            

    def on_rx_done(self):
        print("\nReceived: ")
        self.clear_irq_flags(RxDone=1)
        payload = self.read_payload(nocheck=True)
        print(bytes(payload).decode("utf-8",'ignore'))
        print("\nHumidity: ")
        a = float(bytes(payload).decode("utf-8",'ignore')[:5])
        print(a)
        print("\nTemperature: ")
        b = float(bytes(payload).decode("utf-8",'ignore')[6:11])
        print(b)
        self.set_mode(MODE.SLEEP)

        # Print the values to the serial port
        temperature_c = a
        # temperature_f = temperature_c * (9 / 5) + 32
        humidity = b
        iso = time.ctime()
        # iso = datetime.datetime.utcnow()
        print(
            "[%s] Temp: %sC  Humidity: %s" % (iso, temperature_c, humidity)
        )
        data = [
            {
                "measurement": measurement,
                "tags": {
                    "location": location,
                },
                "time": iso,
                "fields": {
                    "temperature": temperature_c,
                    "humidity": humidity
                }
            }
        ]
        # Send the JSON data to InfluxDB
        client.write_points(data)
        # Wait until it's time to query again...
        time.sleep(interval)

        
##        sleep(.1)
##        print("Send again")
##        self.set_mode(MODE.TX)
##        self.clear_irq_flags(TxDone=1)
##        sys.stdout.flush()
##        self.tx_counter += 1
##        sys.stdout.write("tx #%d" % self.tx_counter)
##        self.write_payload([255, 255, 0, 0, 104, 101, 108, 108, 111])
        
        self.reset_ptr_rx()
        self.set_mode(MODE.RXCONT)
        
##    def on_tx_done(self):
##        print("\nTxDone")
##        print(self.get_irq_flags())
##
##    def on_cad_done(self):
##        print("\non_CadDone")
##        print(self.get_irq_flags())
##
##    def on_rx_timeout(self):
##        print("\non_RxTimeout")
##        print(self.get_irq_flags())
##
##    def on_valid_header(self):
##        print("\non_ValidHeader")
##        print(self.get_irq_flags())
##
##    def on_payload_crc_error(self):
##        print("\non_PayloadCrcError")
##        print(self.get_irq_flags())
##
##    def on_fhss_change_channel(self):
##        print("\non_FhssChangeChannel")
##        print(self.get_irq_flags())

lora = LoRaRcvCont(verbose=False)
lora.set_mode(MODE.STDBY)
lora.set_freq(915.0)
#  Medium Range  Defaults after init are 434.0MHz, Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on 13 dBm

lora.set_pa_config(pa_select=1)

try:
    lora.start()
except KeyboardInterrupt:
    sys.stdout.flush()
    print("")
    sys.stderr.write("KeyboardInterrupt\n")
finally:
    sys.stdout.flush()
    print("")
    lora.set_mode(MODE.SLEEP)
    BOARD.teardown()

thanapol199717, 2021-04-20:
`from time import sleep
from SX127x.LoRa import *
from SX127x.board_config import BOARD
import board
import datetime
import sys
import time
from influxdb import InfluxDBClient

BOARD.setup()
host = "192.168.1.231" # My Ubuntu NUC
port = 8086 # default port
user = "mushroom" # the user/password created for the pi, with write access
password = "mushroom"
dbname = "mushroom" # the database we created earlier
interval = 60 # Sample period in seconds

client = InfluxDBClient(host, port, user, password, dbname)

# think of measurement as a SQL table, it's not...but...
measurement = "mushroom1"
# location will be used as a grouping tag later
location = "Mushroom"

class LoRaRcvCont(LoRa):
    def __init__(self, verbose=False):
        super(LoRaRcvCont, self).__init__(verbose)
        self.set_mode(MODE.SLEEP)
        self.set_dio_mapping([0] * 6)

    def start(self):
        self.reset_ptr_rx()
        
        self.tx_counter = 0
        self.write_payload([255, 255, 0, 0, 104, 101, 108, 108, 111])
        
        self.set_mode(MODE.RXCONT)
        while True:
            sleep(.5)
            rssi_value = self.get_rssi_value()
            status = self.get_modem_status()
            sys.stdout.flush()
            

    def on_rx_done(self):
        print("\nReceived: ")
        self.clear_irq_flags(RxDone=1)
        payload = self.read_payload(nocheck=True)
        print(bytes(payload).decode("utf-8",'ignore'))
        print("\nHumidity: ")
        a = float(bytes(payload).decode("utf-8",'ignore')[:5])
        print(a)
        print("\nTemperature: ")
        b = float(bytes(payload).decode("utf-8",'ignore')[6:11])
        print(b)
        self.set_mode(MODE.SLEEP)

        # Print the values to the serial port
        temperature_c = a
        # temperature_f = temperature_c * (9 / 5) + 32
        humidity = b
        iso = time.ctime()
        # iso = datetime.datetime.utcnow()
        print(
            "[%s] Temp: %sC  Humidity: %s" % (iso, temperature_c, humidity)
        )
        data = [
            {
                "measurement": measurement,
                "tags": {
                    "location": location,
                },
                "time": iso,
                "fields": {
                    "temperature": temperature_c,
                    "humidity": humidity
                }
            }
        ]
        # Send the JSON data to InfluxDB
        client.write_points(data)
        # Wait until it's time to query again...
        time.sleep(interval)

        
        sleep(.1)
        print("Send again")
        self.set_mode(MODE.TX)
        self.clear_irq_flags(TxDone=1)
        sys.stdout.flush()
        self.tx_counter += 1
        sys.stdout.write("tx #%d" % self.tx_counter)
       self.write_payload([255, 255, 0, 0, 104, 101, 108, 108, 111])
        
        self.reset_ptr_rx()
        self.set_mode(MODE.RXCONT)
        
##    def on_tx_done(self):
##        print("\nTxDone")
##        print(self.get_irq_flags())
##
##    def on_cad_done(self):
##        print("\non_CadDone")
##        print(self.get_irq_flags())
##
##    def on_rx_timeout(self):
##        print("\non_RxTimeout")
##        print(self.get_irq_flags())
##
##    def on_valid_header(self):
##        print("\non_ValidHeader")
##        print(self.get_irq_flags())
##
##    def on_payload_crc_error(self):
##        print("\non_PayloadCrcError")
##        print(self.get_irq_flags())
##
##    def on_fhss_change_channel(self):
##        print("\non_FhssChangeChannel")
##        print(self.get_irq_flags())

lora = LoRaRcvCont(verbose=False)
lora.set_mode(MODE.STDBY)
lora.set_freq(915.0)
#  Medium Range  Defaults after init are 434.0MHz, Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on 13 dBm

lora.set_pa_config(pa_select=1)

try:
    lora.start()
except KeyboardInterrupt:
    sys.stdout.flush()
    print("")
    sys.stderr.write("KeyboardInterrupt\n")
finally:
    sys.stdout.flush()
    print("")
    lora.set_mode(MODE.SLEEP)
    BOARD.teardown()
`

thanapol199717, 2021-04-20:
Do you have code to send from arduino to raspberrypi and raspberrypi to arduino? I want to control the device remotely and collect data.

Wei1234c, 2021-04-20:
No, I have no code for Arduino.
I also know nothing of your Arduino code, neither of your LoRa configuration on both ends.

But my code can be used with RPi and ESP8266/ESP32. Messages were sent an received  between them without problems.

closed 2021-04-20 by Wei1234c

thanapol199717, 2021-04-20:
Yes i see you i watch it on your youtube. I will keep trying to do it thanks for your reply to my message. 

thanks you

Wei1234c, 2021-04-20:
The problem may be caused by insufficient power supply.

Transmiiting LoRa module has higher demand of power than supplied, so the transmission failed.