Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #16: Problem with RFM96W version, reports 255

davefes, 2023-04-12:
Thank you for all the effort you put in providing this package.

I bought two "RFM96W" modules from Aliexpress.  The chips are marked
RF96, which might be the reason for my problem.

When I run the program `LoraSender.py` I get an error:
`invalid version` 

So, I modified `sx127x.py` to say:
```py
print(f'version is: {version}')

if version != 0xFF:       
#    if version != 0x12:
    raise Exception('Invalid version.')
```
Appears that the chip version is 255.

Testing LoraSender.py appears to run, at least it says 
it is sending packets.

Testing LoraReceiver.py, runs down to the point where it says:

`LoRa Receiver`

then stalls.

I then decided to run LoRaDumpRegisters.py

I un-commented `def dumpRegisters(self):` in
`sx127x.py` and and on both the sender and receiver I get:
`
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> 
MPY: soft reboot
version is: 255
lora <SX127x object at 3ffe52f0>
0x00: ff
0x01: ff
0x02: ff
...
...
0x7f: ff
`
Does this suggest that the chips I have got are not real chips or
is there something else I need to do?

Appreciate your comments,
Dave

Wei1234c, 2023-04-13:
The basic communication may have gone wrong, so that it only return xFF (255) whatever you read.

davefes, 2023-04-13:
Thank you for the quick response.

I have read through the other issues, ie hard versus soft SPI.

I should have mentioned that I am using a bare-module not a dev board so I will now try a dev board.  

I noticed that you are using some of the strapping-pins (5, 12 and 15) so maybe I need to look at those carefully to confirm that they are setup correctly.

closed 2023-04-13 by davefes