Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #18: Message errors at very low signal levels, CRC errors?

davefes, 2023-04-17:
Testing down at -127dB RSSI, which seems to be the minimum, I notice some errors in the payload.

I am sending a string, for example  '4.70' and at the receiver doing a string to float conversion for further processing.  Sometimes I see a letter appear in the payload, for example '4. v7' and Python's string comparison to '4.70' fails because of the letter.

I have put in error-trapping for this case.  I also  could send a warning when the RSSI goes below -120dB.

I noticed that enable_CRC is False, so will try changing both ends to True.

What do I sacrifice when I set enable_CRC to True?

Wei1234c, 2023-04-18:
It's hard to explain with few words. 

Please refer to the [datasheet](https://www.hoperf.com/data/upload/portal/20190801/RFM96W-V2.0.pdf). :)

davefes, 2023-04-18:
There are several hundred "hits" on CRC in the datasheet.  I was hoping for a summary statement like ... when you enable CRC you reduce through-put by 50% or ??

Thanks anyhow.