Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #19: controller.py AttributeError: 'bound_method' object has no attribute 'low'

davefes, 2023-04-17:
When I tried your suggestion in issue:
https://github.com/Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266/issues/14

> to toggle reset pin (connected with all transceivers):
> controller.reset_pin(controller.pin_reset)
> 
> then transceiver.init()

I get:
```py
File "controller.py", line 120, in reset_pin
AttributeError: 'bound_method' object has no attribute 'low'
```

Wei1234c, 2023-04-18:
Not sure how you came into this. 

You may need to trace `controller.py:line37` -> `controller_esp.py:line49`.

davefes, 2023-04-18:
I came into it by adding those lines to your LoRaReceiver.py example:
```py
controller = config_lora.Controller()

#controller.reset_pin(controller.reset_pin)
#transceiver.init()

lora = controller.add_transceiver(sx127x.SX127x(name = 'LoRa'),
                                      pin_id_ss = config_lora.Controller.PIN_ID_FOR_LORA_SS,
                                      pin_id_RxDone = config_lora.Controller.PIN_ID_FOR_LORA_DIO0)

receive(lora)
```
I am not a Python expert but at line 57 in controller_esp.py I see:
```py
            if in_out == Pin.OUT:
                new_pin.low = lambda : pin.value(0)
                new_pin.high = lambda : pin.value(1)        
            else:
                new_pin.irq = pin.irq 
```
Does controller.py recognise low and high?

Wei1234c, 2023-04-18:
`new_pin` is a Python mock object for attaching the `low` lambda function. 

`pin` is a MicroPython machine.Pin object. 

So, `new_pin.low()` === `pin.value(0)`

davefes, 2023-04-18:
O dear, I should have done a copy/paste!  Thank you for pointing out my error.

closed 2023-04-18 by davefes

davefes, 2023-04-18:
Just to clarify in the example LoRaReceiver.py and you want to reset the SX127x:
```py
controller = config_lora.Controller()
controller.reset_pin(controller.pin_reset)

# transceiver.init() is run when you do controller.add_transceivers()

lora = controller.add_transceiver(sx127x.SX127x(name = 'LoRa'),
                                      pin_id_ss = config_lora.Controller.PIN_ID_FOR_LORA_SS,
                                      pin_id_RxDone = config_lora.Controller.PIN_ID_FOR_LORA_DIO0)

receive(lora)
```