Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #20: ESP32 Deepsleep and CAD mode.

davefes, 2023-05-16:
I would like to put the ESP32 in deepsleep and have it wake-up when there is
any LoRa traffic.  I have read about the CAD mode:

> When in CAD mode, the device will check a given channel to detect LoRa preamble signal

and would like to know if you have tried it or could give some pointers how to
implement it.

Perhaps a new MODE_CAD = 0x07 and then ... any suggestions would be appreciated.

Thank you.

Wei1234c, 2023-05-17:
I didn't try CAD mode.

As you suggested, using DIO to wake up ESP32 may be a good idea. 

However, I have no longer experimental setup on hand, so won't do such modification and tests.

davefes, 2023-05-17:
Thanks anyhow.

closed 2023-05-17 by davefes