Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #21: Waking ESP32 from lightsleep, DIO0 mapping question

davefes, 2024-01-20:
I would like to have the ESP32 controller in lighsleep while waiting for a valid
Lora message.

I seem to be lucky that using DIO0 to both wake-up the ESP32 from lightsleep and
then to process the RxDone IRQ seems to work.

I am concerned that maybe the wake-up should happen before the RXDone interrupt
happens, so I looked at what:

REG_IRQ_FLAGS_MASK = 0x11
REG_IRQ_FLAGS = 0x12
and getIrqFlags() do and it looks like these flags are only used in handleOnReceive(), which I am not using.

Then I thought I try changing the DIOx mapping to give an interrupt on
valid header, ie make a new IRQ_RX_VALID_HEADER = 0x10 (bit 4) and try to
work out how to process it.

Should using DIO0 to wake-up the ESP32 and then process the payload be a 
reliable method?  

Thank you.

Wei1234c, 2024-01-30:
Thanks for your suggestion.

However, I didn't try putting asleep and waking-up the ESP32, so lake the knowledge to give advice. Nor I have plan to modify and test this project further. 

So please fork and try whatever you like on your copy. 

Good luck!

davefes, 2024-01-30:
OK, thanks.

closed 2024-01-30 by davefes