Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266 #22: need explanation

aziaihsk, 2024-09-16:
hello Wei Lin! I'm from china I'm 17 age

I'm so sorry,I came too late,so that i didn't see the content of the URL in the MD file you released ,if you can please send it to me ,or re-maintain the URL,end,Thank You very so much!!😀

aziaihsk, 2024-09-16:
i can give you somethinr in return

Wei1234c, 2024-09-16:
https://wei1234c.blogspot.com/2017/08/sx127x-lora-transceiver-driver-for.html?m=1

closed 2024-09-16 by Wei1234c

aziaihsk, 2024-09-17:
i know chinese,but......it seems that the website can't be accessed?

Wei1234c, 2024-09-17:
> <Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266/issues/22/2355204679@
> github.com>

It is accessable from my phone, no problem.

Maybe China govement blocks it. You may need VPN.

aziaihsk, 2024-09-17:
> […](#)

is so bad，i‘m so distressed